vortexntnu/vortex-acoustics #1: Any updates ?

senceryazici, 2020-03-09:
Hi, I’ve been going through your codes, is it still active ? We have STM32H743ZI and I’m working on it. 

Some questions,

You are using DMA but the sample array is not defined in ram D2 region, your callbacks shouldn’t fire, was that a test code or something ? 

Thanks ..

petterhs, 2020-03-09:
Hi,

This was code for testing. I'm not a part of the team anymore but i can reach out to the team member responsible for the acoustic system this year for you.

senceryazici, 2020-03-09:
Thank you very much, well like I said I'm using STM32H743VIT6, and I was able to get mine working, if no progress achived on this repository, we can share.

We're struggling with Multiple ADC DMA's (not 1 adc multiple channel but instead multiple adc with 1 channel for each). Probably we are doing a stupid mistake, nevermind, I was just surfing and found yours, 

have a good day

Best

chrstrom, 2021-01-11:
Closing due to inactivity (overdue hah)

closed 2021-01-11 by chrstrom