vortexntnu/vortex-acoustics #14: Find a way to communicate with the Xavier

oyssolbo, 2020-12-17:
Important to find/standardize the communication between the STM32 and the Xavier. 
This includes sending data and the time of the transfer, as well as error-logging

senceryazici, 2021-01-11:
Hello I’m from ITU AUV team, I have developed a protocol between the stm32 and the Xavier for use with hydrophones.

Currently, the protocol is able to send configuration messages to configure channels, and the frequency. Also send requests which can start circular acquisition, start oneshot acquisition, stop acquisition, transfer adc buffers, perform FFT and transfer result and couple of more features. Also, all the messages include CRC-16, which is calculated by HCRC peripheral in stm32 without the need of cpu operations.

If you are interested, we can further develop the protocol collaboratively to also suit your needs and our needs in the future perhaps.

have a good day.

oyssolbo, 2021-02-03:
Sorry for the late response, @senceryazici
 
This sounds very interesting. @chrstrom @michoy

michoy, 2021-02-04:
Hi @senceryazici 

Thanks for reaching out. We are interested in working together on this task, but it is on pause for 3-4 weeks due to an upcoming test of a new AUV. We would like to start collaborating after that. Is your protocol available somewhere?

closed 2025-08-15 by forrisdahl