vortexntnu/vortex-acoustics #22: Estimating distance using intensity

oyssolbo, 2021-01-03:
The current estimate uses the power of the sound-source to estimate the distance. It is inadequite, as we often may not know the power of the sound-source beforehand.

oyssolbo, 2021-01-03:
As of 03.01.21, the source-power is given as 177. Which unit is it given in? {W, mW, micro-W, ... }

Areskiko, 2021-01-05:
If you're referring to line 5 in acoustics.py:
[image]
Then it's measured in decibel as per the specifications of the pinger that were found online, searching for the model that was listed in the competition guidelines.

oyssolbo, 2021-01-05:
But how could that be used to estimate the distance. The equation requires power in W to be valid?

Areskiko, 2021-01-05:
I assumed [this](http://hyperphysics.phy-astr.gsu.edu/hbase/Sound/intens.html#c1) meant I could use decibels. If not we need a different approach. Although, even if we can, it's not a very good approach as it's a very rough estimate with many factors that could drastically change the result. Things like echo for instance

oyssolbo, 2021-01-05:
Fair point. I think we can assume that they have used the proper reference-value for dB (10^(-12) W if I remember correctly) and not any other reference giving dBA, dBm etc. 

I am currently looking into the odas-library's predecessor (don't remember its name atm), as it looks like it has some nice trilateration-code

oyssolbo, 2021-01-07:
Forgot to link to the issue before merging. Closing manually

closed 2021-01-07 by oyssolbo