vortexntnu/vortex-acoustics #69: Design bandpass filter bank to isolate pinger pulse

nicbres, 2022-02-11:
*Description:* In Robosub a section of the competition will have its assigned pinger frequency ([check section 9.5](https://robonation.org/app/uploads/sites/4/2021/04/RoboSub-2021-Mission-and-Rules__V1.pdf)). To isolate this pinger frequency and remove unnecessary noise we should use digital bandpass filters.
The idea is to design a filter bank that contains filter coefficients for all the possible center frequencies, to be able to easily switch to new center frequencies.
1. Research filter capabilities of Teensy: e.g., are DSP libraries available, hardware support, etc?
2. Design bandpass filters accordingly. E.g., by using Matlab Filter Designer or similar tools

nicbres, 2022-03-03:
Apparently there is already a filter implementation that was used on the STM https://github.com/vortexntnu/acoustics/blob/master/Source/analyze_data.cpp . I totally forgot about that. You can possibly use this to align with.

ingar-eh, 2022-03-10:
It looks like an IIR-filter, is that something we want to completely avoid? @nicbres

closed 2025-08-15 by forrisdahl